cfry/dde #3: Can't connect to Dexter, "Warning: Dextersim.send doesn't know what to do with the legal oplet: w"

JamesNewton, 2017-05-08:
When I upgraded and try to run my version of forcePlay.js, all I got after pressing any button was 
"Warning: Dextersim.send doesn't know what to do with the legal oplet: w"

It apparently modified my dde_init.js file to the following:
`persistent_set("ROS_URL", "localhost:9090") //required property, but you can edit the value.
persistent_set("default_dexter_ip_address", "192.168.1.142") //required property, but you can edit the value.
persistent_set("default_dexter_port", "50000") //required property, but you can edit the value.
//This file is loaded when you launch DDE.
//Add whatever JavaScript you like to the end
//just before the call to "out" in the last line.

//To change DDE colors,
// 1. Uncomment the below line(s).
// 2. Select the first arg.
// 3. Choose Insert menu "Color" item.
// 4. After inserting the new color, eval the "set_" call.
// 5. To get the default color, just comment out the line and relaunch DDE.
// set_window_frame_background_color("#ff8c96")
// set_pane_header_background_color("#e39f5a")
// set_menu_background_color("#ff8c96")
// set_button_background_color("#ff8c96")

persistent_set("default_dexter_ip_address", "192.168.0.142") //required property but you can edit the value.
persistent_set("default_dexter_port", "50000") //required property, but you can edit the value.
new Dexter({name: "dexter0"}) //dexter0 must be defined.

new Dexter({name: "dexter0"}) //dexter0 must be defined.
`
adding the three new lines at the top:

persistent_set("ROS_URL", "localhost:9090") //required property, but you can edit the value.
persistent_set("default_dexter_ip_address", "192.168.1.142") //required property, but you can edit the value.
persistent_set("default_dexter_port", "50000") //required property, but you can edit the value.

and the extra 
new Dexter({name: "dexter0"}) //dexter0 must be defined.
to the bottom.

This contradicts the following statement in the documentation under "Updating to the latest DDE": "Follow the directions under First Install above. This will not change any files in your dde_apps folder."

JamesNewton, 2017-05-08:
Even after editing the file back to the correct settings, I can no longer connect DDE to Dexter. Dexter is working and I can browse \\192.168.0.142\Share and see all the files there. I can SSH into Dexter. But DDE gives me that same "Warning: Dextersim.send doesn't know what to do with the legal oplet: w" no matter what I do.

cfry, 2017-05-08:
Admittedly this is a terrible error message.
And there's a bug in the simulation software that
I'm working on now.

When you're connected to Dexter and want to run
real (not simulated) instructions,
set the Job's menu item of "Simulate?" to unchecked.

I think using forceplay;js only makes sense in
non-simulated mode.

cfry, 2017-05-08:
You're correct that I am modifying your dde_init.js file
in the new upgrade and that this contradicts the doc.
This code is in play right now.

Here's what I'm trying to do:
Certain features are "configuration" types features
that you mostly want to set and forget.
I'm trying to make it easy to find those features by
putting them in the dde_init.js file with comments
and default values.
For the ip-address in particular, I need to coordinate
better with Kent about how this gets initialized.
If you've got a better one than the new default,
then please use it in the persistent_set call.

There is also the persistent_db.json file
that stores values across DDE sessions.
The problem is that sometimes these values
are set with menu items or other actions the user takes,
and other times they are set explicitly with
calls to persistent_set.
We want to give the user flexibility here,
including set them with code.
Mostly I'm trying to have the user not
directly edit persistent_db.json,
and providing enough help in
dde_init.js to give the user the knowledge to do what they want.
But I want persistent_db.json possible to look at directly
and edit, unlike every other database.

You can get the default dde_init.js by
renaming your dde_init.js (say dde_initX.js)
then relaunching DDE, then
perhaps copy in missing values from dde_initX.js.
This is not good, but these are hard problems
that pretty much no applications get right.

JamesNewton, 2017-05-08:
Argh... I forgot about that setting, and didn't think to recheck menu items after the upgrade.

I guess the real issue is not preserving settings between upgrades, or documenting that they need to be set again.

cfry, 2017-05-08:
Well, I really need to just get stable and not
be screwing around with this stuff,
but in the face of rapid development,
I'm afraid its a bit chaotic.  Sorry.
I *do* appreciate your feedback.
I've added several things to my do-list based on
tonight's email from you.
Thanks.

JamesNewton, 2017-05-08:
No worries. Rapid changes are expected. Emotion has no place in code development, so I should apologize for "Argh..." ,o) It would be great if every code change were accompanied by a doc change, but that slows the pace and I'm pretty much the only one out here at the moment so I don't mind being a little frustrated as you come to stasis. 

I was able to again connect to dexter after unchecking the Jobs / Simulate? option. I dearly hope someone else makes this mistake in the future and finds this closed issue...

closed 2017-05-08 by JamesNewton

cfry, 2017-05-08:
Keeping doc and code consistent is definitely a goal
of mine so please keep telling me when you find something.
Actually I'm impressed that you're actually reading the doc.
I do spend  a lot of time on it.

And its good to have some emotion of frustration.
I want to know about it as I want DDE to be
as less frustrating as we can possible make it.
Keep those great comment coming!