cfry/dde #13: [Linux] desktop file is lacking a Categories= entry

probonopd, 2017-08-18:
The desktop file is lacking a `Categories=` entry in the `*.desktop` file.

According to the [menu spec](https://standards.freedesktop.org/menu-spec/latest/apa.html),

> By including one of the Main Categories in an application's desktop entry file, the application will be ensured that it will show up in a section of the application menu dedicated to this category. If multiple Main Categories are included in a single desktop entry file, the entry may appear more than once in the menu. 

Hence, please add at least one of the following in the `Categories=` key.

Main Category | Description | Notes
-- | -- | --
AudioVideo | Application for presenting, creating, or processing multimedia (audio/video) |  
Audio | An audio application | Desktop entry must include AudioVideo as well
Video | A video application | Desktop entry must include AudioVideo as well
Development | An application for development |  
Education | Educational software |  
Game | A game |  
Graphics | Application for viewing, creating, or processing graphics |  
Network | Network application such as a web browser |  
Office | An office type application |  
Science | Scientific software |  
Settings | Settings applications | Entries may appear in a separate menu or as part of a 		"Control Center"
System | System application, "System Tools" such as say a log viewer or network monitor |  
Utility | Small utility application, "Accessories"

In addition, you could specify one or more from the longer list of [Additional Categories](https://standards.freedesktop.org/menu-spec/latest/apas02.html).

Please test the result with `desktop-file-validate` and make sure it passes.

electron-builder offers native support for this since [v19.22.1](https://github.com/electron-userland/electron-builder/releases/tag/v19.22.1).

Reference: https://github.com/AppImage/AppImageHub/issues/2

probonopd, 2017-09-03:
Latest release is still suffering from this. Should be easy to fix.

cfry, 2017-09-05:
Your email is not sufficient to tell me how to do this.
I have constructed a file named
.desktop
put it in the top level folder of my code,
with the content:

#for linux, required by github
Type=Application
Name=Dexter Development Environment
Exec=dexter_dev_env
Categories=Development,Education,Robotics

Is that sufficient?

After searching for 20 minutes, I found

no documentation on the web as to the

separator character between categories,

so I'm guessing its comma.

This will be in my next release.

probonopd, 2017-09-05:
The specification is at
https://standards.freedesktop.org/menu-spec/latest/apa.html

Did you test your desktop fie using `desktop-file-validate`?

The following passes validation:

```
[Desktop Entry]
Type=Application
Name=Dexter Development Environment
Comment=Develop and run software on the Dexter 5 axis robot arm
Exec=dexter_dev_env
Categories=Education;Robotics
```

cfry, 2017-09-06:
Did you test your desktop fie using desktop-file-validate?
No, I don't know how. I don't even have a linux machine, I do
 test  on Windows and Mac which don't have the notion of
a desktop file.
I see that Categories needs a semicolon between
items.
Thanks!!!

PS: if this is a github requirement, how about if github hosts
the desktop-file-validate tool? Maybe it even runs the tool automatically
and
emails the github author with any errors plus a link
that has documentation (like what's the separator
between categories -: ).

probonopd, 2017-09-06:
Thanks cfry. This is not a GitHub requirement but a Desktop Linux requirement. electron-builder allows you to add a category, and it will write the desktop file for you. Please see the `category` key at https://www.electron.build/configuration/linux-other

cfry, 2017-09-06:
OK, good. So this means I can get rid of
my .desktop file after I've added

"build": {
  "appId": "com.haddington.dde",
  "linux": {
    "category": "Development;Education;Robotics"
  }
}

to my package.json file?

Thanks for your help.

probonopd, 2017-09-06:
I think so, yes.