cfry/dde #14: Job / Cal does not calibrate entire joint.

JamesNewton, 2017-10-09:
Selecting Job / Cal and then selecting at least joint 4 (and I think all joints?) does not actually process the entire rotation of the joint. As a result, errors in the encoder disk, or its alignment, are not seen. Also, changes in the "eye" opening as a result of even minor alignment changes from one end of the rotation to the other are not detected.

This was the cause of the "freak out" in my Dexter arm, which is now resolved.

JamesWigglesworth, 2017-10-09:
Hi James,You are absolutely correct. It's a big problem that the calibration window doesn't allow full disc sweeps.We are currently in the process of updating this calibration window.We haven't actually started any code changes yet so we'd love to hear any criticism of the functionality or aesthetic of the attached design spec. Here are a few notes on the intended functionality:-The user can set the ranges that they want the joint to sweep through with the default being the whole thing-There is a looping check box so the user could set the range over a problem area and scan back and forth while they adjust-The current joint angles will be displayed to determine locations on disc where error occurs.-Multiple joints can be run simultaneously (not critical for this type of calibration)     -Note that all 5 joints can physically be run at the same time but only if J3 rotates in the opposite direction (why its range is flipped)-Instead of five small plot areas all joint eyes are plotted together but with different colors (see legend in bottom right)-Two brand new calibration processes may be added: Auto-Homing and Torque calibration    -Auto-homing will allow user re-zero each joint between power cycles (still in experimental phase)    -Torque calibration measures the stiffness constant of each joint relying on the user to have three known massesThanks,-James Wigglesworth --Haddington DynamicsLead Mechanical Engineer7705 Commercial Way, Suite 125Henderson NV 89011

JamesNewton, 2017-10-09:
The check box for being done with the joint didn't make sense but now that I hear about running all the joints at once, it does again. I would change it to mean include the joint instead of exclude this joint, and have them all checked by default, but that's just a minor thing.

The rest of it makes perfect sense. I look forward to seeing it implemented.

One other suggestion: It might be worth changing the color of the dot after each circle so that you can differentiate one from the next. I'm not sure that has value, other than as an aesthetic.

cfry, 2017-10-10:
"I would change it to mean include the joint instead of exclude this joint,
and have them all checked by default, but that's just a minor thing."

Yes, this is good UI.
_________
"changing the color of the dot"

Yes the displayed dots should correspond to the color of the joint.

JamesNewton, 2017-10-10:
Perhaps each joint could be a different color and every 4 circles could
increase the intensity of the color giving an idea of the part of the disk
that generated the circle. E.g. a dim red is the start of the disk on joint
4 and a bright red is the end.

In my specific case, that would have been very helpful, as the disk was off
at the end, but ok at the start.