cfry/dde #15: Certain Point Sets cause end of DXF file to not be processed

JamesNewton, 2017-10-09:
With some, but not all, sets of points, when processing a DXF file or text_to_lines string, the last several lines or part of a letter are simply not drawn, and the end effector stops while in contact with the drawing plane. To compensate, an "e" or other letter with a curved segment was added for the demo. Other point sets, which wrote at right angles to the original, processed all the data and the end effector was raised above the plane at the end.

The specific point sets which causes this problem were, sadly, lost. I will try to re-create the issue if that will help.

JamesWigglesworth, 2017-10-09:
Interesting...Does it look like the job is stopping before the robot motion is complete?Or are there errors in the inspector?

JamesNewton, 2017-10-09:
I couldn't tell if the job stopped first or the robot stopped first; it seems to happen in both at the same time. No errors... it just appears to toss the last X number of lines.

JamesNewton, 2017-10-11:
In 2.1.7, the following "job" (script) causes the last letter, the "e" to not be written. The arm draws the "-" part of the "e" and then stops, with the pen on the paper. 

    //This job requires DDE version 2.1.4 or higher.
    //var file_path = "choose_file"
    var file_path = txt.string_to_lines("MassMind-org\nJames Newton\nServo Electronicse")
    var height = 0.04;
    DXF.init_drawing({dxf_filepath:  file_path,
    /*
        three_points: [[-.1, .3, height],    //Point1
                        [-.1, .2,  height],   //Point2
                        [-.01, .3, height] //Point3
                        ], //all numbers in meters
    */
        three_points: [[.1, .3, height],    //Point1
                        [.1, .2,  height],   //Point2
                        [.2, .3, height] //Point3
                        ], //all numbers in meters
                        //[x,y,z] x is left right, y is in out, z is up down
        plane_normal_guess: [0, 0, 1], //initial guess of plane normal. (usually [0, 0, 1], [+-1, 0, 0], or [0, -1, 0])
        calc_plane_normal: false, //if true then plane_normal_guess is ignored and plane_normal is determined by three points
        tool_height: 0.057000, //if in home position, vertical distance from the tool tip down to axis of rotation of J4
        tool_length: 0.140000, //if in home position, horizontal distance from the tool tip down to axis of rotation of J4
        DXF_units: undefined,  //units the DXF was drawn in, _mm, _in, _cm etc.
                               //undefined means the drawing will fit the bounding box
        draw_speed: .02,       //cartesian speed in (m/s)
        draw_res: 0.0005,      //resolution for max step size of interpolated straight line in meters
        lift_height: 0.01,     // height that the pen lifts between unconnected lines in meters
        tool_action: false,    ///enable tool actions (true or false)
        tool_action_on_function:  function(){
                                      return [make_ins("w", 64, 2),
                                              Dexter.dummy_move()]},
        tool_action_off_function: function(){
                                      return [make_ins("w", 64, 0),
                                              Dexter.dummy_move()]}
        })

Shortening the string to:

    var file_path = txt.string_to_lines("MassMind-orge")

causes it to finish the "e" and raise.