cfry/dde #16: MIDI sound on Windows 10

JC-Orozco, 2017-11-25:
Is there a forum for this project? I ask since this comment is more appropriate for a forum than an issue report.
In the DDE documentation for MIDI setup on Windows one can find this request "Please tell us if you've had success here and what you did to achieve it." This is what worked for me:
I installed http://coolsoft.altervista.org/en/virtualmidisynth#soundfonts
Then downoaded and extracted "FluidR3 GM Bank" from:
http://www.synthfont.com/soundfonts.html
From VirtualMIDISynth I imported the "FuidR3 GM Bank" sound fonts.
Restarted DDE load a MIDI example code and eval it.

cfry, 2017-11-25:
No good forum yet.
But you in particular should send me direct email just like you did.
Did the above strategy work?
Do the instruments sound natural?
______
Saw you announcement of Makerscad 0.6.
Glad you're still working on this. Haven't played with it
yet but hope to soon.

JC-Orozco, 2017-11-25:
Thanks Fry.
Yes, that strategy worked for me.
I am not a music expert, but the note sounds seemed to be good.

cfry, 2017-11-25:
http://hdrobotic.com/webinar/
has a list of all our video webinar/s one of which is
"Making Music".
Hopefully its good for non-musician programmers
as well a non-programmer musicians.

Conventional music is a horribly unnecessarily confusing language,
the main reason why schools are so poor at music education.
I've tried to help with bringing a little more rationality
to the process.

The school your brought be to in Leon was doing well
with music, but they would be doing MUCH BETTER
if conventional music notation and theory didn't get in the way.
Maybe you could set up a "computer music" program there
to bring some sense to the field. I'll help.

cfry, 2017-11-25:
Such a class could be called
"Intro to programming via music".
Understanding:
- digitizing,
- data structures,
- processes via phrase processing
- formal language
- time & frequency
are all generally applicable concepts.
Acoustics and the psycho-biology of hearing
helps tie this together with other subjects.

cfry, 2017-11-25:
When considering "makerspace" realize
that "making music" is making.
You know I like the idea of making
hardware instruments, but making software
ones is good too.
Imagine building  marimba, then digitizing it!
______
I've just extended DDE's reference manual
with your urls.

JC-Orozco, 2017-11-25:
I completely agree Fry. Software is a very powerful tool, and to make it
accesible is very important goal. Music is a good incentive to learn
programming, since creating sounds and music with programs is fun. Instant
gratification.

Juan Carlos Orozco

closed 2018-01-30 by cfry