cfry/dde #24: Cal saving blank AxisCal.txt files

JamesWigglesworth, 2018-04-05:
If Cal can't read the AxisCal.txt from the robot but can still write to it, 
the file it creates will have all 0's in it except the center for the current joint.

Also a "save as" button should be added for the option to save the file somewhere besides on the robot.

JamesNewton, 2018-04-05:
Suggestion: Update the code to save each joints center point separately (in an array) as the user clicks on it, and only save once ALL are non-zero. Even if read fails, the user can't screw up the settings. 

Suggestion: Automatically save a copy of AdcCenters.txt in the dde_apps folder. If read fails, recover from there with warning. 

Also, updating to the HTTP file transfer is critical so read issues don't happen.

JamesNewton, 2020-07-13:
I believe this was resolved in later versions of DDE. Any reason why we haven't closed the issue?

cfry, 2020-07-13:
grrrr lets bring up at our Software meeting call.
Sounds like we did.

JamesNewton, 2020-07-13:
This should be fixed in the current LTS 2.5.8

closed 2020-07-13 by JamesNewton