cfry/dde #26: Access to the file share on Dexter OS is blocked by Windows 10

JamesNewton, 2018-04-06:
(documenting this here because I don't see it, although I know we are aware of it)

The suggestion has been made to move to an HTTP based file data exchange. This requires the DexRun.c program be updated to support commands for the transfer of file and directory data. E.g. a letter needs to be assigned to a command, and the block of data transferred would be a path or file name as well as a block number since data blocks are of limited size. 

Suggested operations
- Get folder item count: If the command is for a path, and block 0, the returned data would be the number of files and folders under that path. 

- Get folder item data: Then each would be iterated by repeating the path and using block 1, 2, 3... for each folder/file in the path. In that case, the return data would be the file name, size, and any other important properties.

- Get item size: To access a file, first the file path/name would be sent and then block 0 would return the size of the file in blocks

- Get item data: To get the actual file data, the block numbers 1,2,3...  are sent and each returns that block of data from the file.  On the Dexter side, it can be assumed that the file will be read out in block order, so the standard sequential fgets should be fine. Dexter just needs to track the block expected vs the block it's on (by count of calls to fgets with block size buffer) and then send back an error with the current block number if there is an issue. A lost block would, however, require reseeking... I think that can be done with lseek... might be better to close the file after each block and when reading block X, open the file, lseek to that position (block size * block number) and then gets a block. 

- Put item data: Same as before, but it's writing that block of data to the file.

C functions (I think this works in Linux) to get info on a directory:
http://techref.massmind.org/techref/language/ccpp/cref/FUNCTIONS/directory.html

Open a file:
http://techref.massmind.org/techref/language/ccpp/cref/FUNCTIONS/fopen.html

Read data from a file:
http://techref.massmind.org/techref/language/ccpp/cref/FUNCTIONS/fgets.html

Move to a location in a file:
http://www.massmind.org/Techref/language/ccpp/cref/man/lseek.htm
(it may be fseek is right... I haven't done this is a long time)

Write data to a file:
http://www.massmind.org/techref/language/ccpp/cref/man/fputs.htm

If I can get my Dexter setup for development, and this seems good, I'll be happy to try to work on the c code.

JamesNewton, 2018-04-07:
Per CFry:

https://nodejs.org/api/net.html#net_class_net_socket
has some documentation for you.
it LOOKS like we can just write whatever
size string we want.
See 
socket.bufferSize
and 
socket.write(data[, encoding][, callback])

encloding types are
UTF8 and hex as far as I can see in
https://nodejs.org/api/stream.html#stream_readable_setencoding_encoding

JamesNewton, 2018-04-12:
This might be useful for debugging socket connections. e.g. before an oplet is implemented in DDE.
https://jsfiddle.net/vintharas/hbj0emek/

The current list of commands or "oplets" appears to be here:
https://github.com/HaddingtonDynamics/Dexter/blob/master/Firmware/DexRun.c#L2713

Libwebsockets claims to do file transfers. And claims to be light. 
https://github.com/warmcat/libwebsockets
I can't seem to find any examples, but there is this:
https://libwebsockets.org/lws-api-doc-master/html/group__fops.html

I find an example of a file upload via socket at:
https://libwebsockets.org/pipermail/libwebsockets/2016-June/002598.html

Maybe someone can figure out how to implement that?

cfry, 2018-06-21:
The instructions Dexter.read_from_robot and Dexter.write-to_robot
now allow file access without using HTTP.

closed 2018-06-21 by cfry