cfry/dde #29: In Windows 10, suddenly DDE doesn't read dde_init.js

JamesNewton, 2018-05-24:
Didn't change anything, it just suddenly stopped working. It's a real problem because the IP address and all the settings are in there.

luigipacheco, 2018-05-24:
James not sure if its the same problem. But something similar happened to me when we fist used our Dexter, we had to find the right IP with angryipscanner, for somereason it had been changed to what documentation showed.

luigipacheco, 2018-05-24:
James not sure if its the same problem. But something similar happened to
me when we fist used our Dexter, we had to find the right IP with
angryipscanner, for somereason it had been changed to what documentation
showed.

Luis Pacheco
Director
luigi.pacheco@makermex.com+52 (477) 758-40-97 <+52+(477)+758-40-97>
http://www.makermex.com

JamesNewton, 2018-05-24:
No, this is only related to DDE startup, not anything to do with communications with the Dexter robot. I'm understanding Fry already has a solution and actually only raised the issue to provide a place for the answer to be documented.  But it's nice to know someone is reading these... ,o)

cfry, 2018-05-24:
There have been some problems with the ip address of Dexter,
but this particular problem has nothing to do with Dexter,
Just Windows 10 laptops.

      We suspect Microsoft's change to the default setting for a OneDrive configuration
      checkbox in a Windows 10 update in Feb or Mar 2018 causes DDE
      to not launch correctly. The underlying cause is that
      DDE can't find the right folder for its dde_init.js and/or dde_persistent.json
      files. This results in numerous problems like loading and saving files.
      Applications besides DDE have had similar problems.
    
      If you are using Windows 10, we recommend:
      <ol><li>Right-click the OneDrive icon on the Windows Task Bar.</li>
          <li>In the dialog box that comes up, in its right-hand column,
              click "Settings"</li>
          <li>In the next dialog box that comes up, click the "Settings" tab.</li>
          <li> In the "Files on Demand" section, uncheck
               "Save space and download files as you use them"</li>
      </ol>
      If you have additional useful information about this issue, please tell us!

closed 2018-05-24 by cfry