cfry/dde #49: move_to isn't highlighted in the Doc list when searching for move to

JamesNewton, 2020-07-17:
On the Doc panel search box, type in "move to" (no _) and press Find

Drill down to Reference manual, Robot, Dexter, Dexter Instruction. 

Notice that move_to is NOT highlighted. Multiple other instructions starting with move_to like move_to_relative /are/ highlighted.

cfry, 2020-07-17:
Fixed. Now the documentation for the move_to instruction includes the string "move to"
like a number of other instructions do so that it will be found.

JamesNewton, 2020-07-17:
That must be a record for the fastest closing of an issue... LOL.

May I also suggest that the entry for move_to (and several others in that area) start with a basic example of the command? E.g. 
Dexter.move_to([0.2, 0.3, 0.4])

cfry, 2020-07-18:
By convention. examples appear near the end of DDE ref man doc sections.
The move_to doc is unusually long, esp including the picture so
its a bit long to get to the end.
None the less it had NO examples. I've just added two.
Also I've added examples to pid_move_to and pid_move_all_joints, the
only other 2 that had no examples.

You can "almost" see an example in click help. The syntax presented in the output pane is
the "def" syntax, but its very similar to the call syntax, showing you all the arg names and
default values.
I've just made click_help work on the FIND type in to make access easier.
For example: 
Type in Dexter.move_to  in FIND and click on it to see its click_help.
Good bug report!

closed 2020-07-18 by cfry