cfry/dde #90: Verification of recent changes in DDE4

cfry, 2022-08-27:
DDE4 simulator may not behave as well as DDE3.
This task is to verify that DDE4's simulator is up
to DDE3 and to fix incompatibilities.

JamesNewton, 2023-09-05:
I'm still having intermittent problems with the DDE4 simulator. For example, note the J2 slider is at 25 degrees, the job is running, but the simulated arm is still at home:
[image]
here are the errors I get:
[image]

I did wait for the simulated robot to show up before I continued. Note this is behind a certain VPN. On another PC, not using a VPN, the first two sets of errors shows up, but the 3rd one, Cannot read properties of undefined (reading 'rotation') does NOT show up, and the simulator seems to work. So apparently, again, VPN problems. Not sure why...

cfry, 2024-06-07:
fixed

closed 2024-06-07 by cfry