cfry/dde #111: SimObj is_dynamic vibrates excessively.

JamesNewton, 2025-06-17:
Make a stack of objects all set to be dynamic, and watch them vibrate against each other and eventually fall over. 

It turns out that through a happy accident I found a solution to this: Turning on dynamic mode by turning off kinematic mode without changing anything else. e.g. 

```js
var my_obj = SimObj.make_object3d({name: "my_object3d", parent: "scene", geometry: "Box",
    scale: [0.2, 0.2, 0.2], position: [0.0, 0.3, 0.1], orientation: [0.0, 0.0, 0.0],
    material: "MeshNormal", color: [1, 1, 1],
    wireframe: false, is_dynamic: false, mass: 1
})
my_obj.userData.physObj.kinematic = false
```
This gives you an object that is dynamic, but doesn't vibrate excessively.

After looking about a bit in the code, it appears that this routine
https://github.com/cfry/dde/blob/dde4/src/simulator/physicsObject.js#L757
is setting the mass of the object to ZERO! Which, I'm pretty sure, is what makes it shake. Given some mass, they settle down. I'm not sure about the other flags:
- `.setActivationState`
- `.setCollisionFlags`

As those don't seem to be required, but maybe they make the difference? 

More experimentation is needed. E.g. replacing the code for
- `PhysicsObject.makeKinematic()` and
- `‎PhysicsObject.makeDynamic()`
with version which don't change the mass.

JamesNewton, 2025-06-17:
Should try:
```js
PhysicsObject.makeKinematic = function()
    {
        // Tell ammo that this is not a kinematic ovject
        this.rigid_body.setActivationState( PhysicsObject.MotionState.DISABLE_DEACTIVATION );
        this.rigid_body.setCollisionFlags( PhysicsObject.CollisionFlags.CF_KINEMATIC_OBJECT );

        // Set the mass of the object to zero so that it doesn't move
        //this.rigid_body.setMassProps(0, this.zeroVec);

        // Set the velocity of the object to zero so that collisions aren't weird
        this.rigid_body.setLinearVelocity( this.zeroVec)
        this.rigid_body.setAngularVelocity( this.zeroVec)
        this.rigid_body.updateInertiaTensor();

        // The object is now kinematic
        this.kinematic = true;
    }
```

JamesNewton, 2025-06-20:
It's possible the answer is here:
https://n5ro.github.io/aframe-physics-system/AmmoDriver.html#activation-states
with a settings for linearSleepingThreshold and angularSleepingThreshold which allows the object to be "deactivated" while still being a working dynamic object.

JamesNewton, 2025-06-21:
Argh. https://n5ro.github.io/aframe-physics-system is a completely different physics engine that simply has an interface with ammo.js. 

It has been suggested that these settings might help, but so far I see no difference with them:
```js
                o.userData.physObj.rigid_body.setActivationState(1) //alows sleeping
                o.userData.physObj.rigid_body.setSleepingThresholds(3,4) //linear 0.8, angular 1.0
                o.userData.physObj.rigid_body.setFriction(0.9) //0.8 increase
                o.userData.physObj.rigid_body.setRestitution(0.01) //0.1 decrease
                o.userData.physObj.rigid_body.setDamping(0.4, 0.4); //0.1 linearDamping, 0.1 angularDamping
                o.userData.physObj.rigid_body.forceActivationState(4)
```

cfry, 2025-06-21:
Ahh, I was confused by seeing A Frame on the doc
you sent but now it makes more sense,
I did look at AFRAME before using what I did.
I didn't realize AFrame was a wrapper to ammo.
Maybe that would have been a better route.

JamesNewton, 2025-07-07:
Further confusion, errors, and stupidity, I'm sure. Sigh. 

I've attempted to add the following lines
```js
                    debugger
                    if (this.last) {
                    	if ((this.last.p.x - p.x) < 0.0001) p.x = this.last.p.x 
                    	if ((this.last.p.y - p.y) < 0.0001) p.y = this.last.p.y
                    	if ((this.last.p.z - p.z) < 0.0001) p.z = this.last.p.z
                    } else this.last = {p:p, q:q}
                    this.last.p = p
                    this.last.q = q
```

after the line:
https://github.com/cfry/dde/blob/dde4/src/simulator/physicsObject.js#L633
by copying out the entire update routine and replacing it in the editor with
`globalThis.PhysicsObject.update = function() {` 
with the updated code following but that doesn't seem to be having any effect. Even with the Inspect window open, the debugger has no effect, and the source code in the bundle.js file IS still in effect. 

I need to be able to run DDE4 from the source code.